Blockstream/Jade #216: How to save my device from OTA failure?

qdwang, 2025-02-12:
**Device**: LILYGO T-Display-S3

### Background
I purchased the device and flash the Jade latest firmware with secure boot enabled.

Everything works but I forgot to disable the DEBUG log output which could cause the Sparrow wallet failed to unlock the device.

So I try to do the OTA update with the DEBUG log disabled firmware.

The OTA result shows **successful** on both the device screen and my computer's terminal.

### Problem
After I reboot the device, it became this:

[image]

### Is it possible to save my device?
I tried to flash the firmware again by using `idf.py` tools but it failed due to 
```Serial port /dev/cu.usbmodem101
Connecting......................................
/dev/cu.usbmodem101 failed to connect: Failed to connect to ESP32-S3: No serial data received.
```
I think it's caused by the secure boot.

So can I save the device on my own? Or I have to send it to the manufacturer to rewrite the chip?

JamieDriver, 2025-02-13:
Once you enable 'secure boot' you can no longer flash the device.  (That's partly what the 'secure' is about - no-one can take your hw and flash nefarious fw onto it).

qdwang, 2025-02-14:
@JamieDriver 
Thanks for the reply. I managed to recover my device by doing the OTA again with the previous firmware.

It seems the only way to recover is
1. The OTA update functionality still works despite the device's issues.
2. You have a backup of the previous firmware bin file.

su2373466, 2025-03-23:
I had the same problem here, but when I was doing something different. 

I've enabled secure boot and flash encryption, but tried to reinstall the firmware with idf.py bootloader while the device was on without the bootloader mode. Then my device is dead. 

It shows a noise in the screen like an old TV very much like to this one and I don't know if I can resurrect it.

I've tried to upgrade the firmware with python jade-ota.py --noagent but i can't see anything in the ### screen.

There is something I could do or just throw it in the garbage?

Thanks.