Blockstream/Jade #236: DIY Jade on TTGO T-HMI?

zelaya15, 2025-06-14:
I'm not sure if this is where I should post, but I'd like to know if it's possible to install Jade on a Lilygo T-HMI?
I'm not sure which config file to use and if it's compatible.
If anyone could help me with these questions or be so kind as to tell me where I could find more information, I'd be grateful.
I've already searched this repository, forums, and other posts, but haven't found any more information.
Thanks!

JamieDriver, 2025-06-19:
I do not believe so.  Someone would have to setup the config for the device and check the display was supported.

trevarj, 2025-06-22:
Found some info here: https://github.com/Xinyuan-LilyGO/T-HMI

Looks possible, but with some work. (ex. touch controller isn't the same as any supported boards).