Blockstream/Jade #237: Feature Request: Space BTC Addresses Every 4 Characters for Improved Readability

PeterMcBTC, 2025-06-15:
Description:
Currently, BTC addresses on Jade are displayed without spaces, e.g., **bc1q443kwwfu2008vdqgkj727ndpdk2et4m5fvwgld**. This can make them harder to read and prone to errors during verification. Adding a space every 4 characters, as implemented in the Green/Blockstream app, e.g., **bc1q 443k wwfu 2008 vdqg kj72 7ndp dk2e t4m5 fvwg ld**, would improve readability and reduce mistakes.

Proposed Solution:
Update the address display logic to insert a space after every 4 characters for both sending and receiving BTC addresses.

Reference:
The Green/Blockstream app already formats addresses this way, providing a clear example of the desired output.

Impact:
Enhanced user experience with easier-to-read addresses.
Reduced risk of errors when copying or verifying addresses.

Additional Notes:
Please consider this formatting for all address types displayed on Jade to maintain consistency.

JamieDriver, 2025-06-16:
It indeed displays as you suggest on a JadePlus device, which has a larger screen.  

[image]

(The issue with original jade hw is the screen size makes this impractical for some longer addresses (eg. multisig)).

PeterMcBTC, 2025-06-16:
I tested it as a singlesig on Jade Classic and noticed the improvement. I’m pleased the Jade Plus further enhanced readability