Blockstream/Jade #240: DIY - 3D file for Waveshare S3 Touch LCD 2 Touch Display

lukasosterheider, 2025-06-26:
Hi community,

I have a question regarding the DIY version with the Waveshare S3 Touch LCD 2 Touch Display.
There is a linked GitHub page with 3D printing files.

If you know, could you please add to the docs which 3D printing file exactly should be used for this kind/configuration of the Waveshare device utilized for Jade DIY?

I would like to be sure which is the correct one, before I order a case / let someone print the case.

Thanks and best regards
Lukas

trevarj, 2025-08-08:
@lukasosterheider 

Hi, you can [view the STL files on github](https://github.com/trevarj/enclosures/blob/master/waveshare/ESP32-S3-Touch-LCD-2/ESP32-S3-Touch-LCD-2-Rev3-Enclosure%20(Meshed)Buttons%20Lowered.stl) to see what it looks like. When you import the .3mf file into a slicer, it should contain the body and two buttons. If you don't want buttons (using only with USB and never debugging, or OK with disassembling) then you can choose the model without button holes.

I recommend the model with "lowered" buttons. It seems to work well for the way my printer is calibrated.

Let me know if you have any questions, since I created those models and ported Jade to this board.

trevarj, 2025-10-29:
> I modified the case to fit without screws and be able to add a battery

@sandman21vs Nice! Awesome to have a screwless version. 

What is the hole for under the USB-C?

leoperod, 2025-11-26:
> > > I modified the case to fit without screws and be able to add a battery
> > 
> > 
> > [@sandman21vs](https://github.com/sandman21vs) Nice! Awesome to have a screwless version.
> > What is the hole for under the USB-C?
> 
> The battery I used in the project is the official jade battery (single small enough that I had at home yesterday, but I already bought a new one to put in place 602030.

Hi, friend. i know the topic is not about it, but me and some other people are get stuck on splash screen due some camera issues (OV5640). How could you make it work fine on waveshare s3 touch lcd 2?

there is a topic about it https://github.com/Blockstream/Jade/issues/248

thanks already.